monome/crow #54: Confirm I2C addresses to use

trentgill, 2019-02-14:
I arbitrarily chose the i2c addresses as 0x78 through 0x7B.
Couldn't find an exhaustive list of the occupied addresses (though i do remember seeing one). Thinking the choice of i2c address could double as the 'which crow' number when multiple units share a single usb connection.

Perhaps @scanner-darkly has ideas here?

scanner-darkly, 2019-02-14:
yep, 0x78..0x7B range is available. this is the file that is the most current reference for monome i2c address space: https://github.com/monome/libavr32/blob/master/src/ii.h

not the most ideal place for it since not everything uses libavr32 but it's the one that i try to keep up to date, we could move it elsewhere if it makes sense. i can add crow address range once it's stabilized. 

re: using different addresses for multiple units - there are 2 options, one is using a unique address for each unit, the other one is using the same address but passing a unit index as a parameter. if i'm not mistaken, when multiple devices share the same address they all receive commands sent to that address. this could be useful for scenarios where you want to send the same command to all crows (like global reset), although you could just send to each one separately, of course. another benefit of using a parameter is that it allows you to support up to 256 units as opposed to 4.

tehn, 2019-02-14:
0x01-0x0F range is also available! (seems easy to overlook)

@scanner-darkly good idea about same addresses. i like the possibility of having options to do both methods... thinking about some sort of voice clustering. maybe we reserve 4 addresses?

PR #118 (2019-05-23): update i2c address to range starting at 0x01
fixes #54 

0x01-0x04 range selected as per @tehn 's thought. leaves the WR range (0x7*) more open.
closed 2019-05-23 by trentgill