monome/crow #55: Add I2C support for crow to teletype

trentgill, 2019-02-14:
crow is able to be queried by another i2c leader device. teletype is the clear companion device who would lead crow, so the TT project must be updated with hooks for the actions.

This should be implemented for a single crow device first.

Afterwards we can consider how to handle multiple crow devices. This could be sequential input/output numbers (ie `CROW.OUT 5` would refer to the first output of the second device, vs `CROW.2.OUT 1` or `CROW.OUT 2 1` etc).

Proposed initial functions are listed in lua form: https://github.com/trentgill/crow/blob/aa45d96a322f6c818e84a98bc2d69b58a189ed45/lua/ii.lua#L38
and TT form: https://github.com/trentgill/crow/blob/aa45d96a322f6c818e84a98bc2d69b58a189ed45/lua/ii.lua#L67

All the generic functions should send /receive int16 values.

trentgill, 2019-03-01:
to aid integration between crow & TT, a small library of helper functions should be added as a lua file.

purpose is to allow TT familiar users to have a way to treat values in crow the same as they do in TT without worrying about the transformations they hide:
`TT.V(x)` converts a lua volt (ie float where 1.0 is 1V) to a TT volt (ie 1638.4)
`TT.VV(x)` as above but for 163.84
`TT.N(x)` as above but 136.53

note the i2c driver will truncate floating point values, so this lib might need to add 0.5 after confirming the rounding behaviour or TT.

this will be deprecated after #49 is complete.

scanner-darkly, 2020-05-11:
i can help with the teletype side of things once the supported messages are finalized.

trentgill, 2021-06-14:
included in the TT 4.0 beta

closed 2021-06-14 by trentgill