monome/crow #58: allow transmission of long-lua-chunks for immediate execution

trentgill, 2019-02-18:
currently there is a hard limit on the size of an immediate-execution lua chunk sent over TTY.

the limit is set by the `reader[]` buffer size in caw.c, which is itself set by the `USB_RX_BUFFER` define. as of this writing it's 1024 bytes. https://github.com/monome/crow/blob/969b0a952dde7e49fdee06d5e3a7dd19ae32dc84/lib/caw.c#L7

in order to extend this capability we either need to increase the buffer size, or dynamically allocate memory as the buffer reaches the limit.

please discuss.

tehn, 2019-03-01:
trying to understand the use case for this.

i understand that someone might want to send a big function chunk during realtime execution, which is  also quite error-prone compared to writing a script and uploading a big block.

if this functionality is essential i might suggest two new bracketed pre-lua commands `^^queue` and `^^execute` (or something that fits) which would essentially do what flash storage does, but just push it to memory and execute the block when done.

this doesn't strike me as a v1 necessary feature, but please let me know if i'm missing something that relates to immediate needed functionality or an existing bug/problem.

trentgill, 2019-03-01:
I agree it's not v1 at all. Also, 1k chars is already pretty big for a crow script. Let's table it (remove milestone) until it's stopping real-world utility.

ngwese, 2019-10-05:
With the exception of the call to `Lua_reset()` prior to upload it seems like this change https://github.com/monome/crow/pull/193 would largely address the issue. I guess I'm having a hard time thinking of use cases where one would need/want to send such a large body of code to crow outside of swapping out the entire script.

AFAIK the hardware outputs will stay at their current levels over a `Lua_reset()` assuming the incoming script does immediate run an init like function

trentgill, 2019-10-07:
closed by #193

closed 2019-10-07 by trentgill