monome/crow #68: implement different curves for slope library

trentgill, 2019-02-20:
currently all slopes are just linear ramps.
implement the basics with small lookup tables and linear interpolation. even [64] is overkill for sub-audiorate modulation.

once complete, think about how to implement the 'absolute' shapers from `readme-development` & make a new issue.

trentgill, 2019-10-04:
```
-- 2 categories of shapers are provided
--      1. relative shapers (like RC curves)
--      2. absolute shapers (quantizers)
--
-- relative shapers should be simple lookup tables with interpolation.
-- these can be relatively small tables as they are not for audio so
-- distortion isn't much of a concern.
--
-- these will cover the basics:
--  linear (which is just pass through)
--  square
--  log / expo
--  sine / cosine / tan
--  sec / csc
--  1/x
--
-- of note here is that all of these relative shapers for drawing a
-- line between 2 points could be applied absolutely as well, shifting
-- the baseline of the output. consider that these could be different!
--
--
-- the absolute shapers are interesting though as a set of divisions
-- thinking primarily in terms of absolutely dividing up the output
-- space.
-- user provides:
--  zero point to base off (could be a 'key')
--  fold point (1V becomes 1 octave, but could be eg 1v2 for buchla)
--  either | divisions (divides fold-size into n equal steps)
--         | list (divides fold-size unequally according a list)
--
-- the natural use is to provide scales for quantizing. by using the
-- list approach, one can send arbitrary pitches, so this could be
-- tonal scales directly, rather than semitones of 'divisions'.
--
-- list can also take fractions to natively handle just intonation,
-- readily accepting a scale as a list of fractions.
--
-- a weighting mechanism could be added to give a wider window to
-- preferred values.
--
-- setting divisions to a large number, then manipulating fold will
-- essentially become a bit crusher with continuously variable bits
--

local shaper = { zero = 0
               , fold = 1  --volts
               , divs = 12 --semitones
               , list = {} --empty means nil
               }

function S_absolute( samp, divs, fold )
    div_sz  = fold / divs
    bracket = math.floor(samp / fold)
    shelf   = samp % fold
    shelf1  = shelf / fold
    div     = math.floor(0.5 + (shelf1 * divs))
    return ( div * div_sz
             + (bracket * fold)
           )
end

function S_absolute( samp, divs, fold )
    local bracket = math.floor(samp / fold)
    local shelf   = (samp % fold) / fold
    local plateau = 0
    if type(divs) == 'table' then
        if divs[1] == 'ji' then
            --TODO needs log shaping i believe
            plateau = divs[2][math.floor(1.5+(shelf * #divs[2]))] - 1
        else
            plateau = divs[2][math.floor(1.5+(shelf * #divs))] / divs[1]
        end
    else
        plateau = fold / divs * math.floor(0.5 + (shelf * divs))
    end

    --local plateau = type(divs) == 'table'
    --    and divs[math.floor(0.5 + (shelf * #divs))]
    --    or fold / divs * math.floor(0.5 + (shelf * divs))
    return plateau + bracket * fold
end

lydian = {12,{0,2,4,6,7,9,11}}
pentatonic = {12,{0,2,4,7,9}}
just_lydian = {'ji',{ 1/1, 9/8, 4/3, 11/8, 3/2, 10/7, 15/8}}
S_absolute( 0.1
          , lydian
          , 1
          )
```

trentgill, 2019-10-04:
Some `C` notes:
```
// TODO make a multi-dimensional array including:
//  index number
//  string name descriptor
//  function pointer to processor
// then the caller just invokes the set fnptr

#include <math.h> // floorf()
typedef enum{ S_linear
            , S_square
            , S_log
            , S_sine
} S_shape_t;

// Lookup tables
int   LUT_sine_sz = 4;
float LUT_sine[LUT_sine_sz+1] = {0.0, 0.125, 0.25, 0.5, 1.0};
int   LUT_log_sz = 4;
float LUT_log[LUT_log_sz+1] = {0.0, 0.125, 0.25, 0.5, 1.0};

// TODO split this fn into 2, to allow manual inversion / reflection etc
static float S_LUT_interp( float  samp
                         , float  origin
                         , float  dest
                         , int    LUTsize
                         , float* LUT
                         ){
    // find samp's point between origin and dest (0-1.0)
    sinterp = (samp-origin) / dest;
    interp = origin + sinterp * (dest - origin);

    // find appropriate LUT neighbours & make coefficients
    float lut_i = interp * LUTsize;         // eg: 3.3
    int   lut_i_a = (int)lut_i;             //     3
    int   lut_i_b = int_i_a + 1;            //     4
    float lut_i_c = lut_i - (float)lut_i_a; //     0.3

    // pull LUT refs and interpolate
    float a = LUT[lut_i_a];
    float b = LUT[lut_i_b];
    return a + lut_i_c * (b - a);
}

float S_relative( float     samp   // sample value (absolute)
                , S_shape_t type   // choose the type
                , float     origin // zero point for continuous fn
                , float     dest   // end point for cont fn
                ){
    switch type{

    case S_linear:
        return samp;

    case S_square:
        // TODO could add a 'delay' style which always maps to origin
        return dest; // just map to the destination immediately

    case S_log:
        return S_LUT_interp( samp
                           , origin
                           , dest
                           , LUT_log_sz
                           , LUT_log
                           );

    case S_sine:
        // might need an extra step to handle inversion / negatives?
        return S_LUT_interp( samp
                           , origin
                           , dest
                           , LUT_sine_sz
                           , LUT_sine
                           );

    default: return samp; // default to linear
    }
}

float S_absolute( float samp
                , int   divs // 12 // must be >0
                , float fold // 1.0
                //, float zero // assume 0.0 for now
                ){
    // TODO optimize more elements into k-rate
    // k-rate
    float div_sz = fold / (float)divs; // abs size of a div
    // a-rate
    int bracket  = (int)floorf(samp / fold); // confirm this rounds to -ve
    float shelf  = samp % fold; // 0-fold range
    float shelf1 = shelf / fold; // 0-1 range within fold
    int div      = (int)(0.5 + (shelf1 * (float)divs)); // select the div
    return ( (float)div * div_sz
             + ((float)bracket * fold) // add #folds back
           );
}
```

whimsical-sam, 2019-11-14:
https://github.com/whimsical-sam/crow/blob/shapers/lua/shaper.lua

Check out this implementation for user-facing customizable shapers in lua

basic idea: four different domains
- input signal --> convert to phase space based off a foldover value (e.g. 1V, or 1.2V etc.)
- phase space (# of folds + 0-1 phase within a single cycle/octave etc. )
- lookup table (lookup table divides 0-1 into a series of windows; window size can be affected by different weightings, boundaries may be related to the values within the lookup table, e.g. accurately spacing notes in a scale rather than equally dividing up 0-1)
- output space (converting look up table values to output values, e.g. ratios to volts, semitones to volts, or any other custom mapping; interpolation between lookup table values is also availabile; also includes a function mode for using lookup-tables made up of functions instead of numbers)

PR #292 (2020-06-20): output shapers with arbitrary scales
Fixes #68 

The idea here is to implement quantization of the ASL outputs, so that slopes can be seen as sequences of notes. With this extension, the ASL language takes on a new role as describing the 'shape' of melodies (or harmonies), and the absolute-shaper acts as a dynamic quantizer. This is not the final version, but an initial POC to explore what it feels like to play with slopes as melody-shapes.

I'm posting this now because I'm not by my synthesizer very much during the lockdown, and I'd love to see how anyone feels about it in practice. The drivi [... cut]
closed 2020-06-20 by trentgill