monome/crow #82: Hardware test procedure

trentgill, 2019-02-28:
For production units.

What is the simplest way to test all the peripherals. The list:
- Input jacks
- Output jacks
- USB port
- I2C port

Does this require a custom firmware to be uploaded first, or can we make it an automated process?

Note, the firmware is big so takes around 10 seconds to flash. It would be great to be able to physically testing one unit while the next unit is being flashed.

trentgill, 2019-07-07:
1. Attach crow via edge-connector
2. run `make R=1 flash` (could be part of a .sh)
3. while flashing, attach crow to computer as USB device
3a. if flash fails, set aside. PSU, or uC issue.
4. after flash finishes, run `druid/upload.py test.lua` (nb: auto-calibration will occur before this upload)
4a. if error occurs (what should response be?), set module aside for secondary testing: HW issue likely
5. crow broadcasts query over i2c to test module (W/), expects response
5a. if error (no response): i2c issue
6. set 4 outputs to known unique values
7. patch input[1] to output[1:4] -> should light up leds on test module (via i2c)
8. patch output[4] to input[2] -> should flip all leds off on success
9. run `clearscript.py`
TEST COMPLETE

nb: 2 & 4 could be a shell script with a `sleep 1` between steps

``` --- test.lua
function init()
  ii.wtest.get('begin')
  timeout = Metro.init{ event = timedout
    , time = 1.0
    , count = 1
    }
  timeout:start()
end

function timedout()
  print('no response from i2c query. check i2c connection & retry')
end

function test_case( e )
  if e[in_chan] == 1 then
    input[1].mode('stream', 0.1)
    input[2].mode('none')
  else
    input[1].mode('stream', 0.1)
    input[2].mode('none')
  end
  for i=1,4 do
    if i == e[out_chan] then
      output[e[out_chan]].volts = e[expect]
    else
      output[i].volts = 0.0
    end
  end
end

local stage = 1
local expectations =
{ { in_chan = 1, out_chan = 1, expect = -5.0 }
, { in_chan = 1, out_chan = 2, expect = -1.0 }
, { in_chan = 1, out_chan = 3, expect = 3.0 }
, { in_chan = 1, out_chan = 4, expect = 9.0 }
, { in_chan = 2, out_chan = 4, expect = 7.0 }
}
function ii.wtest.event( e, data )
  if e == 'begin' then
    timeout:stop()
    test_case(expectations[stage])
  end
end

local threshold = 0.1 -- should be much tighter tolerance
function input[1].stream( value )
  if value > (expectations[stage] - threshold)
   and value < (expectations[stage] + threshold) then
    stage = stage + 1
    test_case(expectations[stage])
  end
end

function input[2].stream( value )
  if value > (expectations[stage] - threshold)
   and value < (expectations[stage] + threshold) then
    print('TEST SUCCEEDED')
  end
end
```

trentgill, 2019-09-25:
@tehn 
the new druid should be very useful here! it gracefully drops & reconnects to the usb, so if you run it side-by-side with a terminal for flashing, then use druid `r hwtest.lua` to run the test. no need to `^^c` at the end that way.

tehn, 2019-09-26:
https://github.com/monome/crow/issues/186

this throws a wrench in the ii testing.

defining event callbacks doesn't work unless the ii lib is preloaded... i haven't found a quick workaround

trentgill, 2019-09-26:
I thought this already worked, but will accept that. What if you define the event callback inside of init, but after sending a dummy ii command?

Or just send a dummy command immediately before defining, wherever that is?

tehn, 2019-09-26:
tried sending a command immediately prior, didn't help

just running the script twice will do for testing, but the issues persists for in-flash on-boot runs... the script will just fail

moved on the to i/o testing an now struggling with some new lockup issues... trying to identify repo steps and details

tehn, 2019-09-26:
something in the parser is failing for this script:

```
local stage = 1
local expectations = { { in_chan = 1, out_chan = 1, expect = -5.0 } , { in_chan = 1, out_chan = 2, expect = -1.0 } , { in_chan = 1, out_chan = 3, expect = 3.0 } , { in_chan = 1, out_chan = 4, expect = 9.0 } , { in_chan = 2, out_chan = 4, expect = 7.0 } }

function test_case( e )
  print('in[' .. e.in_chan .. '] -> out[' .. e.out_chan .. ']')
  for i=1,2 do
    if i == e.in_chan then
      input[i].mode('stream',0.1)
    else
      input[i].mode('none')
    end
  end
  for i=1,4 do
    if i == e.out_chan then
      output[i].volts = e.expect
    else
      output[i].volts = 0.0
    end
  end
end

local threshold = 0.1 -- should be much tighter tolerance
function is_in_range( value, expectation )
  if value > (expectation - threshold)
   and value < (expectation + threshold) then
    return true
  end
end

input[1].stream = function(value)
  if is_in_range(value, expectations[stage].expect) then
    print('\tok!')
    stage = stage + 1
    test_case(expectations[stage])
  end
end

input[2].stream = function(value)
  if is_in_range(value, expectations[stage].expect) then
    print('TEST SUCCEEDED')
  end
end

function ok()
  test_case(expectations[stage])
end
```

basically anything past the first function will cause a lockup on `r test.lua`

had to reformat the syntax for `input[1].stream = function` as the other way was failing (`function input[1].stream(value)`)

also the `expectations` table was failing with all the extra newlines

---

seems like there are some broader parsing issues, but i haven't tried a really long script yet... curious how long your default.lua script is?

trentgill, 2019-09-26:
are you on osx or linux? i haven't done more than a brief check of druid with osx.

with longer scripts i sometimes have to send the `r <name>` command twice before it won't fail to be parsed (due to some kind of character corruption in the transfer process?)

trentgill, 2019-09-26:
@tehn 
i just ran the above you posted and it uploaded fine & ran ok.
started by just calling `ok()` then followed the prompts until it said `TEST SUCCEEDED` a bunch of times in the druid console.

no issues running the lua script on my end? obviously not the `ii` stuff though.

tehn, 2019-09-26:
running osx, as i suspect that will be our most prominent support target :)

basically `r test.lua` will hard-freeze druid until i disconnect crow

testing on linux now

tehn, 2019-09-26:
works on linux. :P

closed 2019-10-03 by trentgill