monome/crow #93: Need to store semantic version somewhere

trentgill, 2019-03-13:
In automating the binary build `.zip` creation, I need a way to globally access the version of crow.

At present, version is hardcoded as a string in `lua/crowlib.lua` and in `Makefile` itself. These should be co-located somehow, and added to the 'release checklist' which will be required in future.

Perhaps the approach from Teletype could be used, though it uses some Python wizardry I don't understand.

trentgill, 2019-06-07:
Perhaps it could be in a `changelog.md` file and then write a tiny script that looks for the most recent. Then in the PR checklist adding info to changelog with a new version tag would be required.

simonvanderveldt, 2019-06-07:
If you just want the version for building release archives normally one simply tags (with annotated tag normally) a release and then you can use `git describe` to get that version.

The `git describe` part can than be included in the CI config/script that gets triggered on a new tag/release.
It'll take care of compiling everything + do whatever else is necessary and attach the archive to the release here on GitHub.

trentgill, 2019-06-07:
The only other 'requirement' is that a remote host is able to query what version of the firmware crow is running. This means we can auto-detect if a firmware update is available for the attached device.

If we could get that version number built into the binary (as a string?) then it's easy enough to pass into lua. How would you handle that kind of thing? Can we just tack it onto the above git tagging thing?

simonvanderveldt, 2019-06-07:
Once you get the version from git you can do with it whatever you want :)
AFAIK this value then gets written to a header file.

I don't really have experience with Makefiles, I'm of the opinion no one should write those manually :P but stuff like this is easy to do with a decent build tool like CMake or meson.
Something like a combo of https://stackoverflow.com/a/6526533 and http://xit0.org/2013/04/cmake-use-git-branch-and-commit-details-in-project/

There's a pure Makefile example along these lines here https://stackoverflow.com/a/12368262 and there are a couple more of these topics on SO like https://stackoverflow.com/a/6526504.

P. S. Depending on git (the binary) or any other version control when building a project is kind of annoying when combined with github releases, because the automatically generated release archives don't include the `.git` directory. This means it won't work.
The normal way around this is to put the code that gets the git version in a conditional that checks if `.git` exists and potentially another conditional that checks if git itself is installed/the `git` executable is available.

trentgill, 2019-06-08:
note- once this is in, the version needs to be captured by `system_print_version` in ll/system.c

closed 2019-06-10 by trentgill