monome/crow #129: _identify and _version

tehn, 2019-06-07:
minor rework of https://github.com/monome/crow/issues/30

`_version()` should print just the software version (which can be stored as a var/string wherever, doesn't matter)

`_identify()` should return `crow [serial-number]` where the serial number is taken from the STM

on connection, a host will send `_identify()` as text to the tty device and have a timeout period waiting for a response that has "crow" at the front, for correct id as a crow.

trentgill, 2019-06-07:
would it be weird to use `^^identify`? i only ask because it makes implementation easier (i don't have to plumb the serial# through lua). both of these commands seem to be 'remote-only' in that they are meaningless within the crow lua env. they only make sense coming from a remote host, so treating them as commands seems easier.

happy to implement either way, just let me know!

tehn, 2019-06-07:
oh yes that's fine and smart!

PR #132 (2019-06-10): add ^^version and ^^identity queries
also reorganize main.c -> all the stm level config has been pushed into ll/system* to make main.c a better overview of the system's functionality.

## `^^v` (or `^^version`)
returns `^^version('x.x.x')` to be processed as a lua string, where `x.x.x` is the version number (from a git tag).
nb: lua no longer has the `crow.getversion()` function. use `^^v` instead.

fixes #129 

## `^^i` (or `^^identity / identify`)
returns `^^crow_id('0xnnnnnnnnnnnnnnnnnnnnnnnn')` where all the `n`s are the 96bit unique id of the stm chip in hexadecimal. unfortunately the ful [... cut]
closed 2019-06-10 by trentgill