monome/crow #130: possible shortcut vars

tehn, 2019-06-07:
for use as a simple remote, we transmit the string `input` and `output` quite a lot.

perhaps we should create global aliases `i` and `o` respectively? though user error due to redefinition would probably make this a bad decision. thoughts?

trentgill, 2019-06-10:
i'm inclined to leave as is. i don't think throughput is the limiting factor for usb communication and the scripts are already going to be quite terse. collisions w `i` seem likely even though it's rarer than in C. makes me realize i don't know the scoping rules properly in lua, so not sure if an in-function use would trash the alias globally, or just for the duration of a fn call?

perhaps we can consider this a little later on once we've a better picture of use cases?

tehn, 2019-06-10:
agreed!

closed 2019-06-10 by tehn