monome/crow #131: serial reliability

tehn, 2019-06-07:
given we're sending clear text there's no way to check integrity ie via CRC.

on norns sending a stream of

```
output[1].volts=1.234023
```

(where number is different) at a relatively slow tempo (ie 90bpm eighth notes) i end up getting these errors back from crow, roughly one every 3-9 seconds.

```
crow receive: [string "eval"]:1: attempt to index a nil value (global 'oupput')
crow receive: [string "eval"]:1: attempt to index a nil value (global 'ottput')
crow receive: [string "eval"]:1: attempt to index a nil value (global 'outuut')
crow receive: [string "eval"]:1: attempt to index a nil value (global 'oupput')
crow receive: [string "eval"]:1: attempt to index a nil value (global 'ottput')
crow receive: [string "eval"]:1: malformed number near '1..'
crow receive: [string "eval"]:1: attempt to index a nil value (global 'outuut')
```

(but no crashes!)

trentgill, 2019-06-07:
hrmmm. curious that it's generally copying a neighbouring value, rather than dropping it or a random char. i haven't been seeing these issues when blasting (huge numbers of) messages from max, so it makes me think it's something to do with the serial driver.

perhaps something to do with multiple messages being sent within a single usb frame. do you know how the serial output to crow works? currently crow only reads from the usb port every 5ms, but perhaps norns is sending packets every 1ms? i could play with the interrupt frequency to see if it has an effect. or if you want to forge ahead, it's usbd/usbd_cdc_interface.h@70 `CDC_POLLING_INTERVAL`

tehn, 2019-06-14:
changed `CDC_POLLING_INTERVAL` to 2 and 10, neither had an immediately observable effect

placed a dummy print in `caw.c` within `Caw_try_receive` at line 106 so i could see the incoming bytes. sometimes an error is thrown even though the input looks fine. other times the error is visible in the debug UART.

checking the linux driver side (norns stuff) now.

tehn, 2019-06-14:
made a small change to `crow-integration` branch of norns, terminal config bits.

now i'm seeing 100% accuracy printed from the crow UART pre-processing.

tehn, 2019-06-14:
ok this is certainly still something with the matron code.

just did a blast with python (on norns) and it's totally good

tehn, 2019-06-14:
fixed by https://github.com/monome/norns/pull/839/commits/ac63b5e3105e4cebdd2fc210f3f339fe36735409

trentgill, 2019-06-14:
yow! what was the cause?!

tehn, 2019-06-14:
`O_SYNC` terminal setting 😬

tehn, 2019-06-14:
bah. just saw it again. more rare, but still possibly something going on.

tehn, 2019-06-14:
for real fixed https://github.com/monome/norns/commit/cdd5f5a24bd2f12d1795a4b5c4782cf29e5d6d81

closed 2019-06-14 by tehn