monome/crow #147: [crow] .maxhelp and tutorial

trentgill, 2019-07-12:
Need to finish the crow.maxhelp file. Needs to be stripped down to the basics to clearly articulate an interactive model. Currently it's a lot of 'you can do this and this...' but needs to be 'use it this way, so you can do this other thing'.

Talking about things in terms of the modular context, rather than the crow-lua context. The new `.volts` output action with integrated slew time will be a great tool for demonstrating this.

Priorities are:
- set outputs (directly via .volts)
- use the inputs (query, stream, change)
- execute arbitrary lua code (print, variables, functions)
- send i2c commands (with a concrete musical example, eg: playing JF notes from a max MIDI input)
- apply asllib slopes with the `.action` methods

then a separate tutorial in the /rising style which walks through the creation of a complete use-case.

dndrks, 2019-07-15:
this seems to intersect w/the m4l device's core functions, so I'd be happy to help shape this. I'll keep notes as I build.

trentgill, 2019-09-19:
@dndrks i'm closing this as i think your work covers all of this (and so much more!)

closed 2019-09-19 by trentgill