monome/crow #184: output api synchronization

tehn, 2019-09-20:
cc @trentgill 

on norns i did this: https://github.com/monome/norns/blob/crow-integration/lua/core/crow.lua#L110

basically:

```
crow.output[1].receive = function(v) print("output 1 is: "..v) end

crow.output[1].query()
```

`query` sends `get_out` which sends `^^output(1,...)` which i had trigger `receive`

seems like it makes sense to synchronize the standalone api to something similar to this. globals like `get_out` don't sort well into reference table docs

---

pretty sure the input section is sync'd

trentgill, 2019-09-20:
@tehn 
i'm a little confused what the proposal is here?
`get_out()` is really a helper that wraps the output of `output[1].volts` (via __index metamethod) into a `_c.tell()`.

standalone usage won't use the async approach, as you can query `output[1].volts` immediately when running on crow.

...feel like i'm not getting it

tehn, 2019-09-20:
`output[1].volts` is the target voltage

whereas

`get_out()` reports the current position of a possibly slewing voltage or asl

right?

tehn, 2019-09-20:
oh nope i am wrong

trentgill, 2019-09-20:
At present, when you run `output[1].volts = 3` the 3 is sent directly to C as the destination, and is not saved in lua at all. should it be? (and yes, `print( output[1].volts )` is the current value)

tehn, 2019-09-20:
ok so the output query is only relevant to norns because it has to be asynchronous

likewise, input[1].volts notation doesn't make a ton of sense on norns because it also has to be asynchronous.

perhaps i should re-evaluate the naming of some of the norns functions?

closed 2019-09-20 by tehn