monome/crow #197: Ansible (via ii) channels are 0-based

mimetaur, 2019-10-05:
I noticed this today when I connected my new Crow (via Norns) and Ansible. It took me a while to debug why nothing was working, until I happened to notice the channel two light flashing. 

Seems to me everything in Lua should stay 1-based, even if it just means making an offset by 1 in the Ansible definition on the Crow side.

tehn, 2019-10-06:
thanks for catching this. will fix--- it's a bug in the ansible.lua ii definition

ngwese, 2019-10-13:
Ah yes, and I've confirmed just now - this off by one issue was mentioned in #198 as well. That ticket is looking at enhancements to the ii definition system so I'll keep this issue in mind.

trentgill, 2019-11-19:
fixed in #224

closed 2019-11-19 by trentgill