monome/crow #201: Third flash state for a blank slate & initialization choices

trentgill, 2019-10-06:
@tehn @ngwese I want to straighten this question out before we merge #193.

### Prop 1: `Lua_Reset()` should *only* restart the lua VM

Currently this function calls `init()` after restarting the VM, which can cause a CPU & RAM usage spike. This is problematic as it happens right as we start blasting the USB port with the script.

If we do this, `^^k` (which just calls `Lua_Reset()`) won't call `init()` & becomes a quick, non-flash-touching way to get the bare minimum lua env with maximum free memory. If the user wants to start their userscript, they can type `init()` in druid.

Reduces RAM footprint while we're trying to malloc the reception buffer for `^^s`.

### Prop 2: `^^f`(or `^^w`) & `^^e` both call `init()` after a successful transmission.

Currently these don't start the userscript, unless the user calls `^^k` or types `init()` or far worse, adds `init()` to the end of their userscript. The problem with this last point (which i've been doing constantly) is it breaks the above `^^k` 'give me a blank slate' idea, and can result in duplicate metro issues.

///////////////

In `druid` this means that `r` is just a faster version of `u` that doesn't wear out the flash memory.

What is not currently possible (in druid), is sending multi-line blocks of text. There's no way to enter it from druid's command line, but also no way of getting a code-chunk executed on crow in a way that doesn't trash the VM.

Basically we need a method that is like the old `r` which can either upload a file 'on top' of the existing VM state, or perhaps something fancier like uploading a named object (eg function, table, variable). The latter would be more complicated, but allow a richer live-coding experience where you can have multiple chunks of code, and selectively upload them.

Obviously this is way off topic for this issue, but it gives some context around usage of the above i think.

tehn, 2019-10-06:
having `^^k` not call `init()` seems very sensible.

trentgill, 2019-10-06:
@tehn do you think `^^e` etc should call `init()` directly, or does it make sense to do that in the host application? i can't think of a reason you'd use `u` or `r` in druid to upload a full script but then *not* want `init` called.

maybe rephrased: should the end of a successful upload / run sequence act as if crow just restarted? i think the answer should be yes.

tehn, 2019-10-06:
i think that's ok.

upload to flash = reset and run init

pretty intuitive

simonvanderveldt, 2019-10-06:
I don't know if I understand it correctly, but I'd expect the script I upload (`u`) or run (`r`) via druid to be run and work. I guess that means it should run the script in a clean environment?

trentgill, 2019-10-07:
Great i’ll get this PRd today

PR #202 (2019-10-07): Correct initialization of lua env for different upload / boot processes
- default.lua no longer calls init() at the end of the file (crow itself should do this)
- if a userscript doesn't include an init() fn, lua doesn't have an eval error
- using `^^w` to end an upload & 'write' to flash (leaves `^^f/F` free for First)
- any upload (regardless of write-to-flash or run) calls init() after a successful upload.

fixes #201 
fixes #33
closed 2019-10-07 by trentgill