monome/crow #205: Put global functions into (write protected?) tables

trentgill, 2019-10-07:
`time` should be in `util.time`
`tell` should be `crow.tell` or `util.tell`

there's all the C functions (including those 2 above) in libCrow: https://github.com/monome/crow/blob/master/lib/lualink.c#L448

Plus all the functions that are defined in crowlib https://github.com/monome/crow/blob/master/lua/crowlib.lua

pq, 2019-10-07:
lots of global write-protection discussion over on norns...  maybe https://github.com/monome/norns/pull/667 is useful?

tehn, 2019-10-07:
luckily LVM reset happens _all of the time_ with crow, which is good. no attempt to preserve LVM state across scripts.

only a handful of crow things need to be protected, and luckily they are few. would even be sufficient to get them into a table as suggested.

most overwrites will happen by accident, ie `time = 3` since that is a common var

trentgill, 2020-08-02:
Closing as it's a non-issue in terms of user experience.

closed 2020-08-02 by trentgill