monome/crow #209: fast output (re)sets trip up asl

pq, 2019-10-08:
hacking up a monosynth patch and hit this while playing around w/ note retriggers.

here's a distilled test case that repros for me on norns w/ crow `v1.0.0`:

```lua
-- crow stress test

function init()
end

local function sleep(n)
  os.execute("sleep " .. tonumber(n))
end

function redraw()
  print('stressing...')
  sleep(2)
  crow.send('^^v')
  for i=1,100 do
    print('>>')
    crow.output[2].volts = 0
    crow.output[2].action = 'ar(1,5)'
    crow.output[2].execute()
    sleep(0.01)
  end
  print('done')
end
```

salient repl output:

```
...
crow:	[string "local Asl = {}..."]:115: attempt to get length of a nil value (field 'exe')
crow:	no asl active
```

/cc @trentgill @tehn

trentgill, 2019-10-08:
Thanks for the test script. I’ll investigate!

pq, 2019-10-09:
likely related: https://llllllll.co/t/crow-help-general-connectivity-device-qs-ecosystem/25866/95

trentgill, 2019-10-09:
for `exe` to be a nil value it likely means `.action` is not receiving the arguments correctly. i'm guessing this is because of strings getting corrupted as they're sent norns -> crow.

the fact that it seems fine for some seconds to minutes before starting to fail suggests a memory issue. should try calling `print(collectgarbage('count'))` before the `.action` to see if failure is correlated to a higher RAM usage reading.

Dewb, 2019-10-11:
After some reflection I think the `nil value (field 'vooutput')` error is a bigger issue than the repeated `.action` assignment issue, as it results in dropped triggers with no workaround. Does it seem like the two symptoms are related, or should I file a separate issue?

Dewb, 2019-10-11:
The plot thickens.  I did some more testing, and it looks like moving the `crow.output[2].execute()` to the end of `step()`, after the other outputs' `.volts` are set is successful in preventing the `nil value (field 'vooutput')` error. But I still got dropped triggers.

So I took a look at output 2 on an oscilloscope to see if the pulses were not being omitted, or if they were and Plaits was missing them. Surprisingly, the output was not a 0-8V pulse train, but a -5/+5 triangle LFO! (I'm guessing that's the default linear ramp LFO action behavior.) Changing the `.action` set in `init()` seemed to have no effect -- `pulse`, `ar`, whatever, I always got a triangle wave. 

I moved the `.action` back into `step()`, and that made the oscilloscope output look as expected. I figured the `.action` assignment immediately after `reset()` is getting lost somehow, and indeed adding a metro delay of about 680ms in between calling `crow.reset()` and `crow.output[2].action = ...` seems to make the action take effect.

So it looks like there are three interesting things going on here:
* `crow.reset()` prevents scripts from communicating with crow for a period of time while it's removed/re-added. 
* Calling `crow.output[n].volts = x` immediately after a `crow.output[m].execute()` seems to interfere with the execution occasionally.
* Calling `crow.output[n].action = f` too often results in errors (the original issue.)

trentgill, 2019-10-14:
This seems like a race-condition between the slopes library & the usb-input, as it's norns specific. The same script running natively on crow doesn't have the problem.

Probably because either the `.volts` or `.execute` is creating a callback from C into Lua, right after the next command has been parsed & executed over USB.

I'm not entirely sure the best way to go about solving this. It seems the only reason to call `.volts` and `.action` in series is because you want to force the output to a predefined level before starting an envelope / lfo etc. Is the purpose pretty universally to set it to 0V before starting an envelope? If that's the desired behaviour, we can at least provide an additional argument to `ar(attack-time, release-time, reset-to-zero)` to make the issue less present until we can solve it.

pq, 2019-10-14:
my use case was env re-trigger so something like `reset-to-zero` would do it!

trentgill, 2019-10-23:
This was fixed by #238

closed 2019-10-23 by trentgill