monome/crow #210: Suggestion: system functions start with a _

cmcavoy, 2019-10-10:
Have been enjoying reading code examples. I wanted to suggest looking at the lua dsl for Pico-8 for ideas. One pattern they established there was to prepend `_` to any function that's called by the system, so `init` becomes `_init`.

Excerpt from [pico-8 docs](https://www.lexaloffle.com/pico-8.php?page=manual),

```
--------------------------------------------------------------------------------------------
	Program Structure
--------------------------------------------------------------------------------------------

	There are 3 special functions that, if defined by the user, are called during program 
	execution:
	
		_update()
			Called once per update at 30fps
	
		_draw()
			Called once per visible frame
		
		_init()
			Called once on program startup
```

Distinguishing between special functions (though really, isn't every function special?) and normal functions helps, especially for beginning programmers. (IMO)

Take it as you will, really excited to see what people build on the module!

pq, 2019-10-10:
Interesting idea!  The way I've been thinking of the `_` prefix for `norns` is as a way to signal something like privacy which is useful IMO.   Still, this is a great idea and if not that prefix, maybe another?  Conventions like this are super useful and having system lifecycle hooks pop out a bit more would be really welcome.

tehn, 2019-10-10:
for different reasons (because users _shouldn't_ generally play with them) we're moving things into a `_norns` table.

but the `init` function being user-redefined has been presented as a standard use since norns was released, it seems it'd be backwards progress to change it at this point.

appreciate the links to pico8 though!

closed 2019-10-10 by tehn

cmcavoy, 2019-10-10:
> but the `init` function being user-redefined has been presented as a standard use since norns was released, it seems it'd be backwards progress to change it at this point.

Just to clarify, the `_` functions in Pico are meant to be overridden by users, they're just special because the system looks for them and executes them at different points in the loop. Exactly like crows.

Carry on!

[image]