monome/crow #214: Dropped actions with "attempt to index a nil value (field 'vooutput')"

Dewb, 2019-10-11:
Calling `norns.crow.output[n].execute()` on some outputs and `norns.crow.output[m].volts = x` on others in a clock-triggered callback occasionally results in the output action not being executed and this message appearing in maiden:

`crow receive: [string "eval"]:1: attempt to index a nil value (field 'vooutput')`

This is reproducible when some of the `.volts` follow the `.execute()`, but seems to not be reproducible if the `.execute()` is last.

Repro script:
```
-- crow test

local BeatClock = require 'beatclock'
local clk = BeatClock.new()

local function step()
  local a = 4
  local b = 2
  local c = 5
  crow.output[1].volts = a
  -- Calling execute here will occasionally miss actions
  crow.output[2].execute()
  crow.output[3].volts = b
  crow.output[4].volts = c
  -- Calling execute here does not reproduce the error
  -- crow.output[2].execute()
end

function init()
  
  clk.on_step = step
  clk:add_clock_params()

  params:set("bpm", 40)
  params:default()

  crow.reset()
 
  m = metro.init(
    function ()
      crow.output[2].action = "pulse(0.25, 8, 1)"
      clk:start()
    end,
    1,
    1
  );
  m:start()
  
end

function cleanup ()
  clk:stop()
end

```

Dewb, 2019-10-11:
I thought maybe the `write` from norns could be failing to send all bytes in a message (resulting in a partial `volts` and partial `output` combining into `vooutput`?) so I modified `matron/src/device/device_crow.c`:

```c
void dev_crow_send(struct dev_crow *d, const char *line) {
    char s[256];
    strcpy(s,line);
    strcat(s,"\n\0");

    fprintf(stderr,"crow_send: %s\n",line);
    ssize_t bytes = write(d->fd, s, strlen(s));
    if (bytes < 0) {
        fprintf(stderr,"crow_send error: %s\n", strerror(errno));
    }
}
```
With the above script, this eventually results in:

```
crow_send: output[2]()
crow_send: output[3].volts=2
crow_send: output[4].volts=5
crow_send: output[1].volts=4
crow_send: output[2]()
crow_send: output[3].volts=2
crow_send: output[4].volts=5
crow receive: [string "eval"]:1: attempt to index a nil value (field 'vooutput')
```

So it looks like the messages are being sent from norns correctly. I don't see any evidence of `write()` errors.

Dewb, 2019-10-11:
Ah, this may be a duplicate/variant of #131.

Dewb, 2019-10-11:
Checking for partially successful writes didn't reveal anything either.
```c
void dev_crow_send(struct dev_crow *d, const char *line) {
    char s[256];
    strcpy(s,line);
    strcat(s,"\n\0");

    fprintf(stderr,"crow_send: %s\n",line);
    size_t bytesToWrite = strlen(s);
    ssize_t bytesWritten = write(d->fd, s, bytesToWrite);
    if (bytesWritten < 0) {
        fprintf(stderr,"crow_send error: %s\n", strerror(errno));
    } else if ((size_t)bytesWritten < bytesToWrite) {
        fprintf(stderr,"crow_send partial write warning, wrote %d out of %d expected bytes", bytesWritten, bytesToWrite);
    }
}
```

Dewb, 2019-10-13:
Suspect is that the USB rx buffer in crow is hitting capacity. Have a uart connector on order so I can more easily debug crow firmware.

trentgill, 2019-10-14:
Agreed this looks like a USB rx buffer overflow happening on crow. I believe those errors are only reported to the debug port, so makes sense you're not seeing a system level error returned.

One easy test is to add a comment line `crow.send("--")` where you can add more dashes to see if it changes the specific error you're seeing.

trentgill, 2019-10-23:
Fixed by #238

closed 2019-10-23 by trentgill

Dewb, 2019-10-24:
Ran for an hour last night with the fix, no instances of this error. Thanks!