monome/crow #219: output[1].slew = 4.2 causes crash of crow

jlmitch5, 2019-10-13:
Trying to set slew to a higher value (like 4.2), cause crow to crash.  There's nothing printed in the druid console, and ctrl+c and rerun does not work (you have to flash power to crow)

trentgill, 2019-10-14:
I can't reproduce this.
What OS are you using? Are you on crow v1.0.0 and the most recent druid?

trentgill, 2019-10-19:
That single command isn't enough to reproduce the crash.

@jlmitch5 I'm closing this issue now, but please reopen it if you have more info that makes it reproducable!

closed 2019-10-19 by trentgill

jlmitch5, 2019-10-20:
I think you are right, it was in the course of developing my script that I was having issues that seemed to be isolated to setting the slew value higher than 2.1, but now that I'm trying on my final, working script, it seems there is no issue.