monome/crow #220: II jf control has some stability quirks on startup.

jlmitch5, 2019-10-13:
I was having some issues with a script that I’m working on that controls jf via i2c.  It sometimes doesn’t start correctly.  I haven’t quite nailed down the issue, but I believe it is some sort of race condition on load.  Some thoughts:

- jf sent the ii enable message before the system has finished booting up
- a play_note command runs before the ii enablement happens.

When I set my script to run on system power on, it’s like I will hear a handful of the notes, and then it will stop behaving correctly (something like 1n, 2n, 5n, 6n leds will be constantly lit)

I’ve restructured the code of my script to delay i2c enablement and guarded the play note command until it has been enabled, and I believe this has fixed stability.

Will share the script tomorrow, just wanted to get the issue up while I had a second.

trentgill, 2019-10-14:
Could this be that crow is starting up faster than just friends? Seeing a script that reproduces the error would be super helpful.

jlmitch5, 2019-10-20:
Sorry about this taking so long.  I have found a couple ways of reproducing the issue.

If I upload the below script and start up my system.  I get 4 notes of the sequence then JF "crashes" and doesn't respond.  If I run the script after setting jf_mode(1) some other way, the sequence plays as normal.

```
--- midcentury modular motsgso oct 13 19
-- inspired by bowery/shiftregister.lua and mudlogger - kubota
-- patterned shift register sequencer
-- input 1: performance (1st 3rd slow down mangrove, 2nd 3rd speed up mangrove, 3rd 3rd octave up jf)
-- output 1: clock speed modifier cv
-- output 2: mangrove pitch cv
-- ii: 6 stages of ASR via just friends

reg = {}
scale  =  { 0,2,4,5,7,9,11 }
scale2 =  { 2,0,4,7,9,5,11 }
note = 0
scaleChange = 1
numRepetitionsScale = 8 * #scale
oct = -1

function createVoctScale()
    vOct = { }
    for n=1,#scale do
        coin = math.random(1, 2)
        if coin == 1 then
            table.insert(vOct, scale[n] * 1/12)
        else
            table.insert(vOct, scale2[n] * 1/12)
        end
    end
end

q = {}
function init()
    createVoctScale()
    metro[1].event = iiseq
    metro[1].time  = 0.005
    metro[2].event = nextTick
    metro[2].time  = 0.130
    metro[1]:start()
    metro[2]:start()
    ii.jf.mode(1)
end

-- hack to allow 6note polyphony by staggering with metro
function iiseq()
    if #q > 0 then
        ii.jf.play_note(table.remove(q),5)
    end
end

function nextTick()
    if scaleChange == 0 then
        createVoctScale()
    end
    table.remove(reg)
    newVoltVal = vOct[note % #vOct + 1] + oct
    note = note + 1
    scaleChange = (scaleChange + 1) % numRepetitionsScale
    table.insert(reg, 1, newVoltVal)
    for n=1,6 do
        table.insert(q, reg[n]) -- queue a note for i2c transmission
    end
end
```

I've also been able to reproduce the issue with shiftregister.lua by plugging in a clock signal to input 2 on a different case that is already on, so the clock is going before my crow-containing case boots.  I'll get a note or two, then a JF crash.

I also crashed JF by doing ii.jf.get('mode') while shiftregister.lua was running.

closed 2019-11-19 by trentgill