monome/crow #221: Many ii ansible commands overwhelm or partially lockup ansible

ngwese, 2019-10-14:
There are potentially multiple factors at play here. The short version is that calling `ii.ansible.*` functions in rapid succession (such as within a loop) results in commands being partially dropped/ignored at best or ansible partially/fully locking up at worst.

Test setup:
- crow with 1.0.0 firmware; ansible w/ 2.0.0 firmware
- crow ii connected directly to ansible ii (no other devices on the bus

To reproduce:
- power up devices; connect to crow via druid and run:
- `ii.pullup(true)`
- `for i=0,3 do ii.ansible.cv_set(i, (i+1)*2) end`
- `for i=0,3 do ii.ansible.trigger_toggle(i) end`

When setting the cv values in a loop typically only the last cv appears output is set.

When toggling the tr outputs via a loop run from druid the resulting tr state is somewhat random.

In at least one case ansible stopped responding.

trentgill, 2019-10-14:
This is likely because the i2c library isn't queuing commands.
Surprised this is causing ansible to freeze though. Is there a script that makes ansible freeze reliably?

ngwese, 2019-10-15:
I don't have a script which reproduces the ansible freeze. It happened while running the above loops interactively while testing. I'm not 100% certain that ansible was completely frozen but it definitely stopped responding to `ii` messages (until I power cycled it).

csboling, 2019-10-26:
Haven't seen a crash yet in multiple runs of e.g. 
```lua
for k=1,1000 do for j=1,1000 do for i=0,3 do ii.ansible.trigger_toggle(i) end end end
```
with release firmwares. I do have several more modules on the bus, in particular a Teletype (black PCB) is providing pullups. Teletype can also be sending I2C periodically while crow is spamming the bus like this, including TT -> Ansible messages as well as messages to other modules, and nothing has crashed. Will try again later with just crow pullups maybe? It is also possible that this PR https://github.com/monome/ansible/pull/72 to get this libavr32 patch https://github.com/monome/ansible/pull/72 could result in fewer I2C-related Ansible crashes.

trentgill, 2019-11-25:
@ngwese I know you didn't have a clear crash case, but wondering if this problem has been solved with 1.0.2?

ngwese, 2019-11-26:
I was never able to find a consistent repo case but I could repo it.

I’ll experiment with 1.0.2 this weekend if possible.

closed 2019-12-20 by trentgill