monome/crow #233: Upload crashes because of missing \r\n line endings?

simonvanderveldt, 2019-10-20:
See https://github.com/monome/druid/pull/48

Could we fix crow though to either not care about this or just work when there's only a `\n`?

@trentgill do you know why crow expects a `\r\n`?

csboling, 2019-10-20:
It also seems possible that this is getting buffered on the PC side, in pyserial or elsewhere, not sending until it decides it has a "whole line" rather than this being an issue on the crow side.

simonvanderveldt, 2019-10-21:
> I don't see how this would be a crow issue, but if either of you could see how why the parser would fail I'd love to change it on the crow side.

That looks pretty simple indeed. What happens if there are additional characters from that list (lets say something like `\n\n\n`)?

I've just had a look through the pyserial codebase and don't see anything jumping out with regards to changing or messing with the line endings either.
Is there a way I could test this by directly talking to crow (using `screen` or something like that)?

trentgill, 2019-10-21:
In the `\n\n\n` case, crow will match the last `\n` then pass the whole string to `Lua_eval` which will treat it as extra line-breaks afaik. Really seems strange that this issue was causing such havoc, but I'd love to know why.

re: screen. I never got it to work, but i was using `minicom` before brian built the first `druid` prototype.  I remember having to turn on 'local Echo' and either 'Add Carriage Ret' or 'Add linefeed' (don't remember which). Perhaps that has something to do with it.

I'll note that crow explictly adds `\n\r' to every line (see https://github.com/monome/crow/blob/b7876d1f966c9e0e11ee87bfbf0399b8aa53bae9/lib/caw.c#L29) unless you explicitly send a 'raw' or 'error' message (which i don't think is really used). These functions were from a time when we planned to do the host-parsing differently.

If the `druid` change is improving things as much as it seems, then it seems to do with what we're sending *to* crow, so i don't think the `Caw_send*` functions are to blame. But... Worth exploring?

simonvanderveldt, 2019-10-21:
> If the `druid` change is improving things as much as it seems, then it seems to do with what we're sending _to_ crow, so i don't think the `Caw_send*` functions are to blame. But... Worth exploring?

Yeah, I'd like to know for sure/properly understand the cause. Doesn't seem to be coming from crow nor from pyserial :confused:

I'll take another stab at using `minicom`, couldn't get it to work before...

simonvanderveldt, 2019-10-21:
@trentgill are we using no stop bits? :confused: 
https://github.com/monome/crow/blob/7c15d5f0d8da090bfc9f70780c00287a3c023d21/usbd/usbd_cdc_interface.c#L59

trentgill, 2019-10-21:
@simonvanderveldt certainly looks like it! I'll try turning them on here...

trentgill, 2019-10-21:
@simonvanderveldt re: sending `^^` commands, these don't work via minicom. The parser expects the command to be received in a single usb packet. i'd just try sending a `print'hi'` to check loopback.

trentgill, 2019-10-22:
fixed by https://github.com/monome/druid/pull/54

closed 2019-10-22 by trentgill