monome/crow #249: Add a 'delay' function to postpone execution

trentgill, 2019-11-04:
Particularly when using a gate input to sample a voltage input, there can be some lag on the CV side causing pitch droop. This is being noted when using Kria or Teletype to send a 'simultaneous' gate & CV, that the CV has not yet arrived at the destination. There is always going to be some amount of slack in the hardware implementations... so a solution:

The clear way of implementing a 'delay' function is just to use a metro with count=1, so I propose this:
```
function delay( time, action )
  local function devent()
    action()
    metro.free(dmetro.id)
  end
  local dmetro = metro.init( devent, time, 1 )
end
```

This is untested presently.

Thus for a user an existing script is changed like so:
```
-- existing version
input[1].change = function(s)
  ii.jf.play_note( input[2].volts, 5 )
end

-- proposed version, delays the function call by 3ms
input[1].change = function(s)
  delay( 0.003, function() ii.jf.play_note( input[2].volts, 5 ) end )
end

-- proposed version, using a regular named function instead
input[1].change = function(s)
  delay( 0.003, capture_note )
end

function capture_note()
  ii.jf.play_note( input[2].volts, 5 )
end
```

I see a couple of issues worth discussing:
- Using an anonymous function to wrap the delayed body is unfamiliar to many scripters(?)
- Alternatively, using named functions makes this construct far less elegant / terse
- Dynamically assigning metros allows multiple delay timers & named metros, but breaks the ability to use `metro[n]` directly (as the metro.init function doesn't know it is being used)
- In order to `metro.free` the allocated metro the user-fn is wrapped in another function which feels like it could be optimized (how?)
- Queueing up a number of delays could easily use up all the metros (there's only 7 available)

An alternative approach could be to implement this in the C layer using a single hardware timer (reducing lua timers to just 6) and keeping a queue. This would allow many more timers (as many buffers as we're willing to assign, say ~32), and side-steps the issue where `delay` could collide with `metro[n]`. The obvious con of this approach is it's *far* more code. Perhaps this kind of implementation should be considered after first seeing if the lua-implementation is useful / overwhelming the metro system?g

whimsical-sam, 2019-11-04:
Why not have a queue in Lua?  Something like this -

```
queue = {} -- stores tables with an action and a time

-- add a function to the queue
function schedule(time,action)
    -- schedule event relative to scheduler metro's current count
    local destination = scheduler.count+time*1000 
    local event = {}
    event.time = destination
    event.action = action
    table.insert(queue,event)
end

-- callback for scheduler metro
process_queue = function(count)
    for i in 1,#queue do
        local event = queue[i]
        -- if an event's scheduled time has been passed, do it! 
        if event.time <= count then
            event.action()
            queue[i] = nil -- schedules this entry to be removed from queue
        end
    end
    -- remove nil entries from table and compacts the table
    local j=0
    for i=1,#queue do
        if input[i]~=nil then
            j=j+1
            input[j]=input[i]
        end
    end
    for i=j+1,#queue do
        input[i]=nil
    end
end

function init()
  scheduler = metro.init(process_queue,0.001,-1)
end
```

trentgill, 2019-11-04:
This seems quite inefficient to me, but if you want to test it out and see if anything starts getting weird that'd be great. I'm not really sure what will happen when we start to max out the CPU, but likely it will just result in longer times for events to call into the Lua environment (ie. input change/stream, asl-breakpoints, other metros, i2c calls).

The above approach could be implemented in C in a highly efficient way: time-order events at registration time (not in the scheduler), setting the timer to wait until the soonest event (rather than check at a regular schedule).

There are at least 2 more hardware timers available that aren't exposed to Lua (because they are more complex and have a different interface), so we could use one of those to implement the queue.

//

One thing I'd like to push for is that the user should only have to deal with the `delay(time,action)` function (aka sam's `schedule` fn). Like everything else in crow, it shouldn't feel like you're including a library, but just inherent functionality built into the system.This has the implication that a scheduler must be running all the time (or turned off when zero events are queued).

Realistically are we overdesigning this? I've never used more than 1 metro at a time, so perhaps 7 is enough to share between metros and delays? I think using the named metros is far preferable in the general use case so perhaps we should just push that and go with the simple solution?

simonvanderveldt, 2019-11-18:
Why don't we add this delay to all functions? Or is there a reason we wouldn't want to always include a small delay?

It doesn't seem very user friendly to require everyone to wrap every function call in a `delay()` call, right?
When does/doesn't a function need this? And when someone shares a script would it need to be changed depending on the hardware crow is connected to?

simonvanderveldt, 2019-11-18:
@tehn Ah, OK.

I thought because the description started with 
> Particularly when using a gate input to sample a voltage input, there can be some lag on the CV side causing pitch droop. This is being noted when using Kria or Teletype to send a 'simultaneous' [...]

that it was more of a fix/workaround for an issue.

PR #253 (2019-11-18): delay() to postpone execution of a provided function
Uses the metro system, to dynamically create a timer which is thrown away after the action occurs.

Fixes #249 but note the limitations posted in that thread.

I think this is a good interim solution, and would be nice to have access to it in the crowlib. I think the syntax is about as good as we can get.
closed 2019-11-18 by trentgill