monome/crow #259: Auto-enable i2c pullups if first use fails

trentgill, 2019-11-20:
Now that we have an error-message printing when i2c fails due to no pullups, why not automate the turning them on?

The existing functions can remain, and the check will only happen on the first use per-restart of the device.

It’s generally benign to activate them with an already pulled-up bus (when an error happened because of bus-contention / frozen device), so safety is a non-issue.

The way messages are buffered now means we can activate pullups & retry the failed message.

//

We could of course take the alternate approach of just turning them on by default... the resistors are big (40kohms), so you’d need a *lot* of active devices to make it an issue.

Speaking of, i could enable these same weak pullups across jf & w/ modules to reduce the need for powered busses in a system?

It is likely true that the more devices on a bus the more current is needed, so these stacked resistors might be a good solution?

This is uC specific, but i wonder what the pullup resistance is on avr32? I remember the txi/o did this for a while, but the resistance was so small (~1kb?) that it put the current outside of the i2c spec...

trentgill, 2019-11-25:
@tehn do you have feelings about this?

tehn, 2019-11-25:
my thought is just have them on by default since they're weak.

trentgill, 2019-11-25:
Ok. I think this makes sense. I might add something more complex to the typically-follower devices (like w/ & jf) so they only do it conditionally, but let's just roll this into 1.0.2 (along with the potential `delay()` change)

PR #262 (2019-11-25): i2c pullup resistors are enabled by default
fixes #259 
Just enabling i2c pullup resistors be default. I don't think there's any potential for this to cause harm, and it will solve 90% of cases with pullup resistor issues. Better than having to always ask "did you enable pullups"?
closed 2019-11-25 by trentgill