monome/crow #264: ii.kria.help() doesn't show help string correctly

csboling, 2019-11-27:
With crow v1.0.2, in druid I see the following:

```
> ii.kria.help()

> print('****CAW!')
-- commands
ii.kria.preset( number )
ii.kria.pattern( number )
ii.kria.scale( number )
ii.kria.period( time )
ii.kria.position( track, param, pos )
ii.kria.loop_start( track, param, pos )
ii.kria.loop_length( track, param, pos )
ii.kria.reset( track, param )
ii.kria.mute( track, state )
ii.kria.toggle_mute( track )
ii.kria.clock( track )
-- request params
ii.kria.get( 'preset' )
ii.kria.get( 'pattern' )
ii.kria.get( 'scale' )
ii.kria.get( 'period' )
ii.kria.get( 'position', track, param )
ii.kria.get( 'loop_start', track, param )
ii.kria.get( 'loop_length', track, param )
ii.kria.get( 'reset', track )
ii.kria.get( 'mute', track )
ii.kria.get( 'cv', track )
-- then receive
ii.kria.event = function( e, data )
  if e == 'preset' then
    -- handle preset param here
  elseif e == 'pattern' then
  elseif e == 'scale' then
  elseif e == 'period' then
  elseif e == 'position' then
  elseif e == 'loop_start' then
  elseif e == 'loop_length' then
  elseif e == 'reset' then
  elseif e == 'mut****CAW!
```

This only currently affects ii.kria.help because the string is long enough that it can't be sent in a single `Caw_send_luachunk`. Not sure what the preferred way is to handle this in general.

trentgill, 2019-11-27:
Hrmmm... `Caw_send_luachunk` is limited only by the USB TX buffer (currently [1024bytes](https://github.com/monome/crow/blob/d6e0326c5b85c15dbe579a8d002a82b0b0d17728/usbd/usbd_cdc_interface.c#L55)).

The brute force solution is just allocating a bigger buffer (at the expense of runtime memory).

We could add some complex code to break a string into 1024byte chunks [here](https://github.com/monome/crow/blob/d6e0326c5b85c15dbe579a8d002a82b0b0d17728/lib/lualink.c#L338). This would require an arbitrary delay, waiting for the usb buffer to be cleared, or a callback from the usb tx driver, or some kind of queue system. This seems like massive overkill to me.

We could break up the `ii.<module>.help()` into separate parts. Or take an optional string argument like `'commands'` or `'events'` which would only print the part you were interested in.

We could shorten the `.event` helpstring and just provide a single example, requiring the user to write their own. This could be done conditionally at compile time if the string is >1024 bytes.

Other ideas? How small of a use-case is this?

csboling, 2019-11-28:
> We could break up the ii.<module>.help() into separate parts. Or take an optional string argument like 'commands' or 'events' which would only print the part you were interested in.

This sounds like probably the easiest thing - separate sections for commands / getters / event handler example?

trentgill, 2020-03-26:
@csboling i just renamed this issue to reflect the problem so i stop skipping over it bc it says kria :~)

trentgill, 2020-08-02:
I'm closing this one as it feels like too much to be asking of crow. We'd need some drastic restructuring of the serial stream and/or i2c help saving, which feels outside the scope for this small of an issue.

trentgill, 2021-03-02:
Reopening as an extra 1kB of RAM doesn't actually seem like a big problem after further usage.

PR #444 (2021-11-11): Caw module can now stream a const char bigger than the TX buffer
Fixes #264 

Specifically solves usage of `ii.<module>.help()` for modules with a lot of commands (kria, jf, txo, disting etc).

Previously, only the first 1024 bytes would be sent, after which the USB TX buffer would be full & discard the rest of a message. This PR adds a queue system so that longer blocks of chars will be broken up across multiple USB frames.

Note, this solution only works with `const char*` strings as we don't need to copy the data anywhere. It cannot be applied to scripts that generate giant strings.

//

Wh [... cut]
closed 2021-11-11 by trentgill