monome/crow #269: Test framework for lua system

trentgill, 2019-12-04:
Adding a simple set of mocks for the C-interop functions would allow the lua libraries to at least be syntax checked by the build system. At present, if you make a syntax error in a lua file, this won't stop the build and you won't know things are broken until trying to use the library that contains the error.

A simple test script should load the mocks into a lua interpreter and then load each lua library, reporting any syntax errors to the user.

A basic syntax check should be performed on ii descriptors, as it's easy to miss a curly brace if your editor doesn't have linting. The build system will fail in this case, but the error will be called from a code generation script or worse, the C compiler, which makes it difficult to trace back where the issue originates.

trentgill, 2019-12-20:
A basic syntax check can be performed on lua files with `luac` by using the `-p` option (parse only) before compiling. If it fails, `luac` returns an errorcode which can stop `make`.

Could just add this to `*.lua` formula in make.

simonvanderveldt, 2020-07-20:
`luac -p` as a minimum viable check would be good and shouldn't be a lot of effort.
It's not only about what we currently have and if that code is good but more about giving simple and clear feedback to PRs, it's always easy to make a tiny mistake. And this would also be useful for PRs created by users.

trentgill, 2020-07-20:
I just checked and `luac -p` is already run on every lua file that's compiled into the system! If it fails like so:
```lua
luac: lua/crowlib.lua:71: syntax error near 'll'
Makefile:276: recipe for target 'lua/crowlib.lua.h' failed
make: *** [lua/crowlib.lua.h] Error 1
```

Is that enough of a check? Can we close this issue?

simonvanderveldt, 2020-07-20:
Yeah, that looks perfect!

closed 2020-07-20 by simonvanderveldt