monome/crow #271: Timing system for musical timing in Lua & ASL

trentgill, 2019-12-04:
In my use of crow I'm desiring a way to talk about time in a more complex way than just assigning a metro `time`, or setting segment time in `to` calls for ASL.

## Core ideas
- Global tempo control
- Note length notation
- Sequence notation

### Global tempo
Mostly about conventions here. Initially just having a global variable (eg. `_bpm`) which is user-controllable, and used by other timing functions to convert musical-time into seconds.

Secondarily, the ability to switch between internal BPM, CV trigger, other sync sources?

### Note length notation
A set of converters, or a syntax extension(?) that allows time parameters to be written as musical time, rather than seconds directly. These would use the global BPM to calculate their actual time (at runtime preferably).

Reference MaxMSP's `n` notation:
* `1n` = 1 bar = semibreve
* `16n` = 1/16 bar = semiquaver
* `8nd` = 3/16 bar = dotted quaver
* `4nt` = 1/6 bar = triplet crochet

### Sequence notation
Presently it's difficult to reason *when* something will happen in the future in a musical context. Even after having the above note-length-notation, you need to add up a sequence of times to know when something will occur.

Instead, it seems preferable to use bars-beats notation, though perhaps there is a better solution that doesn't rely so heavily on a fixed time-signature.

Ultimately, there could be some kind of table-syntax so that a composition could be written out in a table as pairs of timestamps & actions:
```lua
sequence = seq.new
{ { 1:1, reset_sequencer(), chord(1) }
, { 2:1, arpeggiate( {3,4,7}, 4n)
, { 2:2, output[3] }
, { 4:1, self.restart }
}
sequence:start(1:1)
```
This would need a lot of additional layers to make work, but the fundamental idea is being able to describe events in time relative to an origin (ie. `2:1` is 1bar after the origin).

junklight, 2020-05-05:
"other sync sources" - it would be nice to have native sync to the Norns clock - I'm just setting up a bit of a set & thinking about making some kind of gate pattern generator on the Norns - If I could easily sync it to the Norns clock (which is sync'ing in turn to Ableton via link) that would be lovely :-)

trentgill, 2020-05-05:
@junklight indeed I imagine norns would be the primary 'other sync source' in consideration.

Do you have any ideas about how you'd want to use it? What should a crow script look like that is synced to norns?

We could likely implement the `clock` system from norns, leveraging the internal timers, then create some kind of clocking mechanism. Perhaps we can use `link` messages directly, so long as it doesn't require too massive a library on the hardware. Would be amazing if that could happen directly from a computer too. @junklight would this make things easy to do what you're desiring?

junklight, 2020-05-05:
I've just put the new Norns clock with the co-routine in Kria and it was really nice - I guess something that had a similar API would be ideal: 

nornsclock.run( pulse) 

function pulse()
  nornsclock.sync(1/8) 
  do_something() 
end

P0ed, 2020-05-06:
There is need for some clock data type, which can be divided/multiplied, assigned to output and perform arbitrary function on clock. So should be the way to populate those clocks from Norns, onboard input or even by calling a some sort of `advance` function.

trentgill, 2020-05-06:
@P0ed I don't think I understand you completely. If the new `clock` system were available on crow, would that give you everything you need?

I'm looking at the `clock` system on norns now and it should be easy enough to get the module running on crow. The main unknown is how to synchronize the clocks in the background (though users should never have to worry about this, or know how it works).

P0ed, 2020-05-06:
Not exactly, `norns.clock` syncs to global clock / tempo from within, crow doesn't have controls to define master clock. So there is two options left:
• set tempo and call start/reset at some point
• synchronize to external clock (input gates stream, norns, `ii`, midi)
It would be nice to have ability to sync one `clock` to another, not just global, which allows to build unrelated clocks.

> The main unknown is how to synchronize the clocks in the background

I guess it's the most difficult part:)

trentgill, 2021-06-17:
closing this as we now have `clock` working which satisfies the first 2 initial desires of this issue (global tempo, note-length notation).

of note there is no current method to sync to a usb-host (eg norns), but there is a new issue to discuss that #404.

the last part about 'sequence notation' is a nice thing to pursue, but there's no clear design proposition here. it's the kind of library that would be prototyped as a crow userscript, so feels fine to let go of it from here, and consider adding a library in future once someone comes up with that idea & executes it.

closed 2021-06-17 by trentgill