monome/crow #305: Input mode setter should take optional 'event' param

trentgill, 2020-04-19:
At present, the user is required to do something like:
```lua
input[1].mode( 'stream', 0.1 )
input[1].stream = ...

-- or table syntax:
input[1]{ mode = 'stream'
        , time = 0.1
        }
input[1].stream = ...
```

Propose to use the 'event' style from the Metro library so the event can be assigned as part of the configuration
```lua
input[1]{ mode  = 'stream'
        , time  = 0.1
        , event = ...
        }
```

trentgill, 2020-06-19:
Closing this as it's already available with a little trick. When you use the input table call you can set *any* parameters of the input table, and this includes the event handlers for each mode.

```lua
input[1]{ mode  = 'stream'
        , time = 0.1
        , stream = function(v) print(v) end
        }
```

@tehn It would be more idiomatic to wrap the above behaviour behind a virtual table element `event`, and have that auto-redirected to the current mode, but i'm not sure if it's worth the extra code? Wouldn't be too difficult.

closed 2020-06-19 by trentgill

tehn, 2020-06-19:
i think the "little trick" is sufficient and looks good