monome/crow #318: Update CI to build the release package

trentgill, 2020-05-06:
At present I believe the CI implementation just runs `make`.

It would be preferable if it build `make R=1 zip` which sets 'release mode' (ie disables debug prints), and builds the zip file with the scripts for uploading via `dfu-util`.

Benefits:
- Less experienced users would have an easier time installing bleeding-edge versions (thus more pre-release testing/feedback)
- The installed version is more efficient & has more free RAM as it doesn't manage a separate buffer & UART.

@simonvanderveldt @csboling : Is this just the `ENTRYPOINT` in `Dockerfile`? If so I can update this & test.

csboling, 2020-05-06:
I think the GitHub actions already build R=1 when you create a release and can upload artifacts in that case as well, configured [here](https://github.com/monome/crow/blob/dae66cb675422f5590ef74a7da34fefead163050/.github/actions/build-firmware/action.yml#L7) and [here](https://github.com/monome/crow/blob/dae66cb675422f5590ef74a7da34fefead163050/.github/workflows/build-release.yml#L18)-ish. I dunno that there's a way to upload artifacts to github without making a release but you could make "v1.0.4-rc001"-style prereleases maybe.

trentgill, 2020-07-08:
Fixed by #346

closed 2020-07-08 by trentgill