monome/crow #323: Using more then 3 logarithmic shapers freezes crow

sbl, 2020-06-12:
I can reproducible get crow to "freeze" (becomes unresponsive) with the below

```lua
function init()
  for i = 1,4 do
    -- notice logarithmic shape
    output[i].action = ar(0.1, 0.1, 10, 'logarithmic')
  end

  input[1].mode = 'change'
  input[1].change = function(_)
    print('tick')
    for i = 1,4 do
      output[i](true)
    end
  end
end
```

By changing this to only 3 outputs (`for i = 1,3 do output[i]() end`) this is not freezing.

I verified this with all shaper modes and only `logarithmic` hangs up.

trentgill, 2020-06-19:
@sbl I just ran the above script with no issue on the current main firmware. If you could test it out using the binary from [here](https://github.com/monome/crow/suites/815455371/artifacts/8936696) that'd be great!

sbl, 2020-06-20:
hi @trentgill thanks for this! just tried out the new binary and I can confirm that this works without issues.

closed 2020-06-20 by sbl