monome/crow #331: Add 'cleanup' function support to scripts

trentgill, 2020-06-20:
Add a 'cleanup' style function to scripts so that code can be executed when 1) the script is overwritten, 2) crow is reset.

This isn't quite the same meaning as on Norns because we build a whole new lua environment whenever a new script is uploaded.

Should the function be called before or after the new upload has been detected?

The primary reason for this is so scripts that change state on remote modules via ii, can unset that state before exiting. Specifically `ii.jf.mode(state)` messages that mean a JF-synth script leaves JF in synth mode even if the new script doesn't have control for it.

//

Counter argument: This should just be the users responsibility to do manually via druid, or max, or norns. It's a one-liner that's not too difficult to remember.

The danger of having synth mode unset automatically on script change is that it will cause JF's oscillators to all start running which could be quite abrasive.

What do we do if a revised version of the script, or a new script that also uses jf.mode(1) is uploaded? If the cleanup function were implemented the module would momentarily switch to normal mode, then back to synth mode which could cause a loud burst of harmonically unrelated noise when the script is loaded. This feels worse than solving the problem we're trying to solve no?

tehn, 2020-06-22:
i'm feeling like the cleanup method would be used rarely? particularly given script updating happens via druid interactively--- so there's access to manually setting the mode--- and as you point out i feel there's isn't a "reasonably assumed" default state when moving between scripts.

though this might need to be tackled in the case of a more willy-nilly script-selector if implemented on norns or m4l w/o the interactive element of being able to correct an unintended jf mode. hmm.

i'm inclined not to fix a problem until it arises. not sure cleanup is a problem yet. good thing to consider though!

trentgill, 2020-06-22:
I agree! I opened the issue because i’ve gotten at least 3 emails from people reporting that ‘jf doesnt work’ or ‘jf is stuck in synthesis mode’ after using a script that has it.

It seems there is an expectation that a script that magically turns on synthesis mode would turn it off on completion.

Perhaps in a script-selector context (which includes jf-synthesis/geode patches), it could be idiomatic to always set `ii.jf.mode(s)` to either 0 or 1 at the beginnig of the script. That way we dont have to add a whole new functionality, but can ensure a known third-party state when needed.

tehn, 2020-06-22:
emails, guh, cripes. ok, this makes sense!!!

trentgill, 2020-07-07:
closing this one, as it seems our current situation is ok!

closed 2020-07-07 by trentgill