monome/crow #340: USB-MIDI device support

trentgill, 2020-06-23:
Crow should simultaneously appear as a serial device, and a USB MIDI device to a connected host. This will allow out-of-box usage of crow with any MIDI capable host, specifically making integration into DAWs other than Ableton possible.

The full scripting environment will still be used, so the precise behaviour of the MIDI interop will be reconfigurable.

Defaults will of course be MIDI->CV conversion, but also MIDI->I2C bus conversion (eg. play Just Friends from your DAW, or sequence a hardware synth with the input jacks.

With #339 this could allow auto-synchronization between a USB host and crow's `clock` module.

trentgill, 2020-07-15:
@tehn to write a super basic script example of what the syntax could look like / what features would be enabled.

tehn, 2020-07-16:
norns midi reference: https://github.com/monome/norns/blob/main/lua/core/midi.lua

basically the same, without `vports`. `to_msg` and `to_data` are key. ie:

```
function midi.note_on(note, vel, ch)
  midi.send{type="note_on", note=note, vel=vel, ch=ch or 1}
end
```

where `midi.send` breaks it down into bytes and forwards to HID-MIDI endpoint.

## midi output

```lua
--- play a random note on in 1 change
note = 0
input[1].mode("change")
input[1].change = function(state)
  if state==1 then
    note = math.random(127)
    midi.note_on(note)
  else
    midi.note_off(note)
  end
end
```

## midi input

`midi.event` gets called with midi data, which is first passed through `to_msg` for clarity. `to_msg` should have a `raw` type for match failure.

```lua
midi.event = function(e)
  if e.type == "note_on" then
    output[1].volts = midi_to_volt(e.note)
    output[2]() -- bang envelope
  elseif e.type == "cc" then
    if e.cc == 10 then
      output[3].volts = 10/e.value
    end
  end
end
```

trentgill, 2021-11-08:
Consider #404 alternate clock sources if this ever makes it. would be nice to support MIDI input as the clock source, and feels preferable to a custom solution with lua functions etc.

Could be used with Norns quite easily as the MIDI clock system is built-in, rather than having to add custom params for turning on a special 'crow' sync.

dndrks, 2024-07-08:
@trentgill , someone reached out to monome support after seeing [your recent post on this topic](https://llllllll.co/t/crow-midi-over-usb/46771/10) asking about feasibility -- work on this hasn't been slotted / prioritized yet, right?

trentgill, 2024-07-09:
@dndrks i don't have anything to add beyond what i wrote on that post. i've made a proof of concept, but it would require some extensive work to get it all built out and rolled into a new version.

if this does get done, it will happen when we have some spare cycles (rare). if someone wants it sooner, they'll need to do the leg work themselves which is no small feat. i won't promise it will happen as there may be showstopper issues as yet undiscovered.

tehn, 2026-04-18:
@mechanical-dance reading the last comment in will illuminate the situation