monome/crow #350: (meta) crow issue forgiveness

trentgill, 2020-07-19:
In the spirit of https://github.com/monome/teletype/issues/206 it seems a good time to reflect on the state of the issues list now that we've launched 2.0

Here's the roundup from @tehn 
> many are stale feature requests and propositions--- many of initiated by myself--- which are largely no longer relevant as the system has shifted dramatically in some ways.
>
> the norns crew recently did a massive purge of stale/won't-fix issues and it felt amazing.
>
> i'm proposing something similar here, and since we're not co-located (and drinking beer), i think it'd be good to weigh in on specific issues prior to removal... though also perhaps there isn't much to discuss?
>
> i'm going to use this issue as a thread to list possible issues to close. any comments are welcome. after some period or time i'll execute the agreed closures.

trentgill, 2020-07-19:
reverse chrono

* #5 - ii-strings are too much work for little reward. rest is complete already
* #11 - seems not to have been issue in practice (as the lua system is restarted on every boot / script load)
* #40 - little room left for improvement. favour #112.
* #67 - stale. probably overkill.
* #81 - i've tried many times and can't solve. small subset of affected users (small portion of linux users).
* #112 - close in favour of #301 
* #125 - close in favour of #301 
* #205 - non-issue (similar to #5 )
* #264 - feels like we're asking too much of crow? should this be a druid / norns feature? just need better documentation?
* #268 - they work fine as-is.
* #269 - most libs are pretty stable. feels unimportant except (maybe) for new additions.
* #279 - likely fixed since 2.0
* #285 - prefer #339 for this edge case
* #302 - input libs are really nice. crow is not designed to be a DSP platform
* #322 - implement as a userlib post #339 

CC @tehn @dndrks @csboling @simonvanderveldt @pq @ngwese @nordseele

simonvanderveldt, 2020-07-20:
Isn't #205 a duplicate of #11 instead of #5?

And I've added some comments to the issues that I understand :P

trentgill, 2020-07-20:
Indeed I got some of the cross-linking wrong. Updated above. Of note, #40 should be closed in favour of #122 (not 112), and our discussion should be copied there.

tehn, 2020-07-20:
#81 is the only one i'd like to take another look at. would appreciate help from @simonvanderveldt (the linux guru!)

all others i am totally good to close

trentgill, 2020-08-02:
I closed 13 issues from above! Left #81 as requested, and also #125 to remind myself to make this a part of ASL2. Closing this issue now.

closed 2020-08-02 by trentgill