monome/crow #364: Support for fractional ii type

trentgill, 2021-01-23:
Currently Disting EX's ii support requires users to set parameters in a range from (0..16383) or (0..65535), I'm not sure which is appropriate.

Currently we have `s16V` for allowing crow to represent 'volts' as multiples of `1.0`. This is 'crow native' representation of volts (ie, how we describe output voltages). The ii driver maps `1.0` to `1638.3`, and casts to int16_t for transmission as a 16bit integer over i2c. This is the expected form for Teletype which de-facto defined the ii specification.

I would love to hide the complexities of integer-ranges for different datatypes from crow scripters. There is no reason for most folks to need to remember the value of 2^16, and instead they just want a logical min & max.

//

I propose we go through the list of supported `ii` devices and note any commands that have magic-number ranges and look for a unified way to represent them. I would suggest `[0..1]` as the arbitrary unipolar range, and `[-1..1]` for bipolar.

In particular I'm thinking of ER301 & Disting EX as interfaces that use lots of these (i think).

cc @CarlColglazier

trentgill, 2021-06-17:
@scanner-darkly wondering if you have a few moments to give a brief rundown of how integer ranges are used in the `EX` teletype system? do the meanings of the integer values change depending on mode?

the above discussion is perhaps not totally related, but just thought you might have some insight on how best to represent the values in crow? when you use the `EX` ops on TT, are you just typing in raw integers, or do you use `V` or `VV` etc?

scanner-darkly, 2021-06-17:
@trentgill this is what we came up with for teletype/disting specification:

for notes, it uses the teletype convention (14 bit for 0..10V range) for both volume and pitch (and for pitch it assumes 1V/oct so 0..16383 translates into 10 octaves). so you would use the same values for volume/pitch you would send to just friends.

for algorithm parameters, there are 2 options:
- set a parameter to the exact value passed - this is useful for when you know the exact value you want: `EX.P 7 23` will set parameter 7 value to 23.
- set a parameter using 0..16384 range which will get mapped to parameter min..max range - this is useful for when you want to, say, map the param knob to a parameter: `EX.PV PARAM`. 

does that answer your question? happy to provide more info!
here is the spec: https://github.com/scanner-darkly/teletype/wiki/DISTING-EX-I2C-SPECIFICATION

trentgill, 2021-06-18:
@scanner-darkly
This is perfect & seems like a nice way to support i2c messages that have multiple contexts / meanings. I’ll have to think on how best to integrate w the automated crow bindings, but it is a nice direction.

Also seems like it would apply well to JF’s Geode mode which uses the same i2c cmds & synth mode, but with small integers, not voltages. Much food for thought!

trentgill, 2021-11-08:
closing this as it's totally aspirational and feels like a solution looking for a problem.

closed 2021-11-08 by trentgill