monome/crow #385: Event queue overflow doesn't show error

trentgill, 2021-04-19:
Currently, if a script causes the event queue to overflow, it fails silently, potentially leading to strange errors.

There is an error raised, but it only prints to the UART debugger.

Adding a call to `Caw_send_raw()` will alert the scripter that their script is creating more events than can be handled.

tehn, 2021-04-19:
does this ever potentially happen when it's not a user script design problem?

i'm wondering if crow will become semi-unresponsive with a full queue, and if it makes sense to on overflow clear the queue and alert the user that their script is busted. (maybe even abort the script?)

trentgill, 2021-04-19:
i believe it's always a user script design problem. i don't think it will become unresponsive, but it will discard events until there's room to store them. i'm honestly not too worried what happens when the queue overflows, but just want there to be an error message so the user *knows* the script is busted (and can't expect anything to work properly).

The line in question is here https://github.com/monome/crow/blob/0f2e2bfbc34d8002b979c9aefa35648fac283a0a/lib/events.c#L79

Oh, but I guess you're point is about whether it would create issues in getting back to an ok state. To which I think the answer is we don't have to worry - for every cycle through the main-loop, the TTY input handler is run once, so live-typed `crow.reset()` or `^^reset` or `^^clear` from the REPL will remain responsive, meaning it's easy enough to recover from the bad state.

tehn, 2021-04-19:
perfect, sounds good!

trentgill, 2021-06-15:
fixed in #391

PR #391 (2021-04-22): Event queue overflow error messages
Just a basic "event queue overflow" message added in case the userscript causes the event queue to blowup.

Fixes #385
closed 2021-06-15 by trentgill