monome/crow #413: delay can conflict with metro

trentgill, 2021-06-24:
at present, `delay` is implemented in terms of `metro`. it allocates a metro, then de-allocates itself once the event occurs.

this causes issues if a script uses the `metro[n]` method of directly accessing metros. this is because an indexed metro call isn't "allocated", and so `delay` doesn't know not to use that particular instance.

rather than trying to fix this in the metro lib (which i'm skeptical is possible), i propose re-implementing `delay` in terms of `clock`:

```lua
-- untested!
function delay(action, time, repeats)
	local r = repeats or 0
	return clock.run(function()
			for i=1,1+r do
				clock.sleep(time)
				action(i)
			end
		end)
end
```

an added benefit is it enables up to 100 simultaneous delays, as opposed to being limited by metro's 8 instances.

this would technically be a breaking change because the `delay` function currently returns a `metro` reference, while now it returns a `clock` coro_id. i think this usage was very rare (most uses just discard the return value), but just calling it out as a potential issue.

tehn, 2021-06-24:
clock is the preferred way of doing delay in norns, so i'm all for this
change

trentgill, 2021-07-03:
fixed in #416

PR #416 (2021-07-02): Delay with clock
fixes #413 

tested & works the same!
closed 2021-07-03 by trentgill