monome/crow #418: Broken norns scripts with 3.0

trentgill, 2021-07-01:
We deprecated / moved some norns functions and they have broken a handful of scripts. @tehn made a list of these somewhere (i think in the norns crow-namespace PR) & they should all be one-liner fixes.

alternatively, i'm not totally opposed to keeping some old aliases to these functions. ideally we would do both, while ensuring the documentation is clear about which to use. that way norns scripts won't break on older-versions of norns (and scripts that dont have PRs merged / aren't public) won't be broken.

dndrks, 2021-07-01:
list: https://github.com/monome/norns/pull/1362#issuecomment-866941444

cheat codes doesn't actually have any breakage, from what i can tell, so safe to skip it :)

trentgill, 2021-07-06:
closing as we did a bunch of PRs to fix breaking changes, and merged a couple backward-compatibility changes into `main`

closed 2021-07-06 by trentgill