monome/crow #453: ASL is running at half speed

trentgill, 2021-11-16:
This was a bug introduced in 3.0

Should be a simple fix, but i'm more worried why it's happening. Need to validate 48kHz sample rate etc.

daveriedstra, 2022-06-25:
Hi, just ran into this manifesting as slews on outputs are taking roughly 2x the expected duration to complete. Verified with a timer at 5s / 30s / 1m. Crow v 3.0.1. Happy to do more testing if that would help!

joeweiss, 2022-10-14:
Would this be something in the lua files, or something deeper? I might be able to take a stab if someone can point me in the right direction.

trentgill, 2022-10-14:
it's very likely some change in C. It's either an issue in `lib/slopes.c` where timing values are being multiplied by 2, or it could be that the DSP loop has been slowed down by half. One helpful thing would be to measure the output samplerate. This is easily seen on a scope by setting a very fast lfo/oscillator on the output, and looking at the pacing of the staircase'd output.

trentgill, 2023-02-22:
Confirmed today that the DAC itself is running at 24kHz when it should be 48kHz. I believe this regression happened when we went to v3.0 but i'm struggling to find a culprit at that point in time. Going to do some quick tests to check that the system clocks are running at their correct speed, and see if there's anything else running at the wrong frequency.

PR #472 (2023-02-22): fix samplerate being halfspeed
Fixes #453 

I'm not sure why this has happened which gives me pause about the solution. What I'm doing now is just increasing the frequency the I2S driver runs at by 2x. Measured samplerate at approximately 48kHz on the scope, and slope times are now accurate compared to the samplerate.

I tried to increase the incoming PLL so the constants were correct in the library, but it seems the PLL is close to maxed out already. Uncertainty why, but this still solves the problem. I've checked that Metro, Clock and Input.stream modes are all running at the correct speeds, [... cut]
closed 2023-02-22 by trentgill