monome/crow #454: `input[n].mode = 'clock'` is undocumented

trentgill, 2021-11-18:
should appear [here](https://monome.org/docs/crow/reference/#input-modes) but there is no reference.

trentgill, 2023-02-03:
from #464 

* add documentation for changing `ii` address (`ii.address = _`)
* improve documentation for `unique_id` function [not sure what this means]

trentgill, 2023-02-03:
other new features already in `main`:

* `ii` "high-reliability" mode #457
* `sequins` string mode #446
* `math.rand` vs `math.srand` #442
* `sequins` current value inspect #436

trentgill, 2023-02-03:
all the currently open PRs that are soon to be merged need docs added to the main page. most of them have extensive usage & examples in the PRs, but need to be distilled into documentation. it probably makes sense to capture them in some kind of blog-post style introduction -- perhaps linked to my website or a lil stub page on the monome site? unsure where that stuff should live.

dndrks, 2023-02-23:
raaaad, these are all amazing additions to an already-fantastic toolkit!!

re:
>[...] perhaps linked to my website or a lil stub page on the monome site? unsure where that stuff should live.

i'd love to keep these on the monome site with the rest of the scripting docs, if you're okay with it? also totally happy to build supplemental materials and just to links out to your write-ups, but since folks will likely email help@monome.org with clarifying q's over their ownership, it'd be nice to not have to task you with updates + addendums.

trentgill, 2023-02-23:
Yeah i'd be totally happy for them to live on the monome docs site. Mostly just conscious that the conversational style isn't quite aligned with the rest of the docs. Very happy if you want to edit them into something a little more straightforward / with less assumptions :)

trentgill, 2023-03-08:
Closing this as all the *info* is now in the monome docs. there's still more to do with editing / tidying up / re-organization, but the docs are now up-to-date with the 4.0 release, so let's start a new issue if there's specific issues that arise.

closed 2023-03-08 by trentgill