monome/crow #460: 'make -j' build successful, but with errors

ninayanez, 2022-01-10:
_*this isn't urgent,  and I'm not sure if it needs an issue._

I built a **crow.bin** from the main branch so I could use `ii.address` to make it easier passing data between crows (there are 3 daisy chained on ii bus). 

`make -j` seems to run successfuly and builds **crow.bin** but output has these errors:
```
lib/lualink.c:32:10: fatal error: lua/bootstrap.lua.h: No such file or directory
   32 | #include "lua/bootstrap.lua.h" // MUST LOAD THIS MANUALLY FIRST
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
lib/ii.c:8:10: fatal error: ../build/ii_c_layer.h: No such file or directory
    8 | #include "../build/ii_c_layer.h" // GENERATED BY BUILD PROCESS
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
I used **osx_linux-update_firmware.command** to flash each crow and functionality seems to be good, they all talk over ii and with jf too. 

I don't work with C and embedded stuff much, so i'm not sure if the error output is worth worrying about?

trentgill, 2022-01-12:
thanks for the comment! appreciate your feedback, and if you have any other experiences to share about the build process i'd love to hear it. we don't have a lot of people building from source, so your feedback is super useful.

this is something i've tried to fix numerous times but have not been successful. it has to do with the way the dependency files are built in Make, but nothing i've tried to make the dependency tree clearer has helped. the files it complains about are built by lua scripts in the build process.

i'm closing the issue for now as it doesn't stop the build from succeeding and unless we have picked up a Make wizard in the last few months i'm not sure it will be fixed...

closed 2022-01-12 by trentgill

ninayanez, 2022-01-12:
cool, the build process was good! I was worried about the errors because things went smoothly otherwise and typing `make` is sometimes like slipping into an arcane underworld (for me at least). I guess I'm not used to the main branch of projects building and running, and was braced for more pain. So far so good.  
Anyhow working with crow and the lua libraries has been really fun, I keep getting lost in it (in the best of ways).