monome/crow #474: v1.1 hardware changes

trentgill, 2023-02-22:
for discussion of implementation of the 2 small changes for the hardware in 2023.

## hardware-pullups for i2c bus
current interface from the reference:
```lua
ii.pullup( state ) -- turns on (true) or off (false) the hardware i2c pullups. on by default
```
original hardware has ~40k resistance applied ("weak" hereafter)
new hardware has option for ~3k3 resistance ("strong" hereafter)

2 options that i see:
1) maintain same API, but new hardware always uses the hardware pullups instead of the uC internal resistors
2) add support for a string argument to `ii.pullup` where we could use `"weak"`, `"strong"`, or `"off"`.

if we choose the 2nd option, should the `true` behaviour be the same as original hardware, or switch to the hardware pullups? and also, should the default be `"weak"` or `"strong"`?

## status LED for debugging

i think @dndrks had some thoughts on what we're trying to communicate. i believe our intention is to have this LED to help us debug issues people experience in the field where they aren't sure if the system is booting up at all.

the led is a single colour, but we can a) flash it, or b) dim it with PWM.

things that we could communicate:
* is microcontroller is running?
* is a USB device connected?
* is any part of the code in an error state? specifically DAC, ADC, USB
* have we entered a Hard Fault state, ie. crow has crashed. (i'm not certain we can reliably do this, but i can try). This would need to be solid on/off, as the uC stops executing.

what else do we want to communicate? what should the dim / blink pattern be for each state?

NOTE: if we want to indicate bootloader mode separately, that must be changed in the bootloader before flashing the hardware in-house.

dndrks, 2023-02-22:
re: LED, the primary support issues are:
- is crow receiving power?
- is crow successfully running a script / otherwise working fine (eg. not crashed)?
- has my USB port failed?

microcontroller status also seems useful for us to be able to identify from an LED pattern.

DAC state seems easy to diagnose at home, as lots of folks write in with stuff like "I've got 10V on every output", so that seems less ambiguous.

does that help?

tehn, 2023-02-22:
@trentgill my preference (not strong) is for option A, default "on" (/strong) --- which covers the predominate use case of norns dongle driving JF. people w/ TT will be able to navigate this territory

trentgill, 2023-02-22:
@tehn agreed that option A is better. doesn't change the API at all & simply provides a more reasonable amount of pullup current.

@dndrks ok i think we can do all 3. how about this:
1) is crow receiving power? turn on the LED at the very start of the stm program. So if the LED comes on, we know the stm has booted which means we have the power running successfully. even if the program quickly crashes, the LED should at least blink when turning on the case.
2) crow running normally: slow fade in/out (never turns all the way off). around 2seconds cycle time.
3) crow running & connected to a USB host: quick fade cycle (~500ms cycle time).
3b) alternatively we could have the same as (2) but flash the LED fullbright whenever a message is sent/received over USB. Would this be helpful beyond "USB connected"? eg. for cases where druid isn't working, but crow hardware is working ok (is this a real-world problem case?)

@dndrks do we need to indicate that we're in bootloader mode? if so, i need to make a change to the bootloader project (which is separate from this repo).

---

i'll get the hardware stubs built in now, but i don't have new hardware to test. should be pretty basic though, and i'm confident i can get it working without hardware in hand. if we have issues getting things up, i'll have you send one of the new units over.

dndrks, 2023-02-22:
@trentgill , those proposals all sounds good!

1. yeah, having any indication that the module is on / able to turn on would help cast aside doubt when crow shows as disconnected in druid for other reasons.
2. normal / standalone 2s cycling makes sense!
3. mirroring data sent/received over USB would be rad! even if just for non-druid situations (eg. Max or norns) where you want quick visual confirmation that the message you sent was received by crow, so then if you aren't getting the result you want it's likely a code issue.

i think indicating bootloader would be a nice-to-have, since druid announces that crow is disconnected when crow _is_ fully functional but just happens to be in bootloader vs. when crow isn't actually connected.

lmk if there's anything else i can help with! <333

tehn, 2023-02-22:
let us not not not touch the bootloader. i've already flashed these all and want to just ^^b them with the final firmware. i'd also like to avoid spending too much time addressing edge cases which benefit _only_ this newest version pcb

i do like 1+2. effectively indicates POWER and NOT-CRASHED

how can we make #2 lowest possible cpu use?

if 3 is the packet flash, let's just have it be an XOR so it's one line without any timers or anything, which i think is a cpu burden for a 99.9% unused feature

trentgill, 2023-02-22:
ok i'll get working on this. already got the pullups done & ready to test.

re: 3) i note that the LED is not visible when the module is screwed into the case. i don't love the idea of telling people to remove the module from the case to check the status light when running, but i guess it's a nice to have for last-ditch-effort debugging before claiming a hardware issue?

i'll take a look at the bootloader - should really just be copying the same code into the bootloader. perhaps in BL mode it can just blink twice a second so it's a distinctly different visual.

@tehn i think it should be possible for druid to detect crow in bootloader mode. this would be a very helpful thing to communicate when entering bootloader mode i think! but it could be more of a burden than necessary. perhaps a better solution is to just match on `^^b` and print a message saying "crow is now in bootloader mode, open another terminal and run `druid firmware` to update crow's firmware". would probably be a 10 minute change?

---

after @tehn's comments:
ok, noted re: CPU usage. it also drastically simplifies the code to just blink rather than worry about PWM. updated proposal:
1) turn on LED at boot (if LED is stuck on, it suggests some kind of uC setup failure)
2) without USB connection: toggle high/low once per second for a 2s duration squarewave
3) when USB is connected: toggle high/low 4x per second (for 500ms squarewave), and XOR the output state on USB tx/rx.

i think this effectively communicates the 2 working states, and should give some blinky / half-bright fuzziness when sending USB packets. it will also be super minimal CPU usage - simply adding a condition check in each background loop, then when it's time to toggle, it just sets variable & toggles the GPIO.

dndrks, 2023-02-22:
the super minimal version sounds right to me!
also, having druid reflect bootloader state would be super super helpful, and would meet the need for any LED indication.

trentgill, 2023-02-23:
There's a dev branch firmware zip over here: https://github.com/monome/crow/suites/11149257223/artifacts/568918905
It includes these v1.1HW changes & the timeline library.
Hope that helps @dndrks !

PR #475 (2023-03-01): v1.1 hardware updates
fixes #474 

I2C pullups should be running now. same syntax as before: `ii.pullup(true/false)`.

For testing on hardware, please measure the resistance from each i2c pin to the +3v3 test point in both on/off states. should be under 3k3 for on, and more like >100k for off. Not entirely sure the resistance measurement will while it's in circuit though... If that's the case, then you'll need a scope. Hook it up to a Just Friends & send `ii.jf.play_note(0,1)` in a fast clock, and probe the i2c pins. When on, the digital signal should be dramatically more square than when off  [... cut]
closed 2023-03-01 by trentgill