monome/crow #483: v4.0.2RC fails when accessing `input` in druid

beels, 2023-05-13:
First a couple of notes on the release: Like others, when I install relaase 4.0.2 (via `fledge` or via `dfu-util`) crow reports in druid that it is running v4.0.1-14-g7721314, not v4.0.2.  I note that github has a 4.0.2RC tag, but no actual 4.0.2 tag.  4.0.2RC contains a commit that does bytecode cross-compilation, but does *not* contain the i2c fixes in commit 34166ec.

The issue I see is: if I connect to crow running v4.0.1-14-etc via druid I get the following response if I try to access the inputs:

```
//// druid. q to quit. h for help

 <crow connected>
 <listening at ws://localhost:6666>
> print(input[1].volts)
?:-1: attempt to call a nil value (field 'io_get_input')
stack traceback:
  ?: in function <?:44>
  (...tail calls...)
  repl:1: in main chunk
runtime error.
```

If I do not try to interact with the DAC, things seem to work fine.  I haven't attempted to use any of the v4 functionality, though.

This crow is serial number 0x002e00133756501420383558.

I am running druid 1.1.4 under python 3.8 on Windows.

beels, 2023-05-13:
Note that if I check out v3.0.1, I get similar error messages about missing files in `build/`, but the compilation succeeds and the resulting binary works fine when installed on crow.

discohead, 2023-05-14:
same issue here. downgrading to v4.0.1 makes this go away.

trentgill, 2023-07-24:
fixed in 4.0.4

closed 2023-07-24 by trentgill