monome/crow #490: Add support for i2c2midi to firmware

eethann, 2023-10-21:
It would be so fabulous to be able to use i2c2midi + crow in standalone mode as a generative CV/i2c to MIDI bridge, especially integrating midi controllers into scripts. Is there room to incorporate the initial work done on this by @mreid (https://github.com/mreid/crow-i2c2midi/blob/main/i2c2midi.lua).

I'd love to help jam on this though it might take a long time for me to get the compile environment set up at the moment.

mreid, 2023-10-21:
I’d love to see this too. I have a more memory efficient and fully featured version of that script you linked to on my machine. I’m traveling at the moment but will push an update once I’m home next week.

eethann, 2023-11-06:
I just uploaded a compile of the first rough implementation and it seems to be working. Yay. I'll work on smoothing it out a bit and maybe getting more coverage of the i2c2midi commands.

mreid, 2023-11-06:
Nice work! I've just pushed the version I mentioned above to my repo: https://github.com/mreid/crow-i2c2midi/blob/main/i2c2midi-compact.lua

From memory, it gets rid of the range checking of the original to save memory and adds some extra commands. It also has a simple example usage at the top.

eethann, 2023-11-06:
Awesome. I just tried with the added CC out ops and they work. So fun!

eethann, 2023-11-07:
I figured out the CC getter (have to pass both channel and control number as args) and got the midi getters basically working. I've published a release so others can try it out here: https://github.com/eethann/crow/releases/tag/4.1.0-i2c2midi-rc1

I haven't gotten to any documentation yet. There's one problem that might require a change to some deeper code in the crow firmware: the second arg is used to pass the control number to query, but the event callback function is only passed the first argument. This makes it impossible to handle a specific controller number in the callback. The solutions I can see are either

  1. Generate separate getters for every CC controller number, so you only pass the channel (seems less than ideal, but would follow the fader implementation)
  2. Update the callback handler to pass an args table or an `arg2` property back to the controller so we can check the specific controller that is being reported on (I think this would be ideal if not prohibitive)

Interested in any thoughts or recommendations about those two.