monome/crow #491: tuning issues

daveriedstra, 2023-11-05:
I was getting my feet wet with crow's frequency tracking and oscillating capabilities and noticed that they're both slightly out of tune. For instance, if I oscillate output[4] at 440Hz, my tuner shows it as about 50c flat. If I then take that output, patch it into input[1], and track its frequency, crow reports roughly 449Hz. (Interestingly, these deviations roughly cancel out, eg, I can use crow to oscillate at a frequency that it tracks from input[1] and it will be pretty close.)

Possibly related to #473

trentgill, 2024-01-16:
i think this is somewhat expected due to the inaccuracy of the internal crystal that crow clocks based on. we don't have a low-speed clock on the circuit board, so we'll always have +/-3% range to deal with.

this offset shouldn't drift, so we could null it out relatively easily. would require an extra calibration step where you feed a precise frequency oscillation to input[1] and measure the difference.

as you note, the input & output should cancel each other out. this is because the output oscillation are affected in exactly the same way. so.. at least it's internally consistent?

daveriedstra, 2024-01-16:
Gotcha, thanks for explaining! I never thought through the fact that clock requirements would be different for a microcontroller and a tuner.

I think tuning crow is completely reasonable, especially if it can be done once during a calibration routine. Players can use the pitch pipe output from a tuner or metronome (through some input module, probably) as the pitch reference.

closed 2026-11-15 by trentgill